sparkfun/SparkFun_RTK_Firmware #731: Logging SD card icon change by single bar

caradag, 2023-11-05:
Hi, I have a problem with the logging icon (SD card with a C inside in my case, because of custom configuration), changes to a single vertical bar. What that means? The GNSS still writes a file but why is there a bar instead of a logging icon?
In the user manual, there is no description of what the bar means.

I'm using Facet firmware 3.9 and nothing is connected to it.

Thanks

nseidle, 2023-11-06:
Welcome!

The vertical bar indicates that no logging is happening, but that the system is running (it's there to indicate if the system has frozen). When logging is active, you'll see the C, P, or lines inside the page icon. 

If the log file size does not increase after 5 seconds the card is marked as offline and the vertical bar is shown. 

How long does your Facet write to the card before it marks it as offline?

Things to try:

* Reformat the card with the [SD org formatter](https://www.sdcard.org/downloads/formatter/). 
* Try a different card / different size. RTK products support up to 32GB.

tonycanike, 2023-11-06:
I see the vertical bar when I forget to put the card in! :)

caradag, 2023-11-06:
Thanks for the advice. That is much clearer now.
After restart, the normal logging icon came back. I'll keep an eye to find out how to reproduce the problem of the bar showing up after a period of normal operation.
Cheers

closed 2023-11-07 by nseidle