sparkfun/SparkFun_RTK_Firmware #732: Can't copy/paste Lat/Lon/Alt coordinates into Serial Menu Base Geodetic coordinates

tonycanike, 2023-11-24:
### Subject of the issue
I can't paste lat, long, height (aka altitude) into the Serial Menu Base Geodetic coordinates.

I have the coordinates in NotePad but I can't copy/paste them into the serial menu.

I can type them in manually, but who has time for that!?

### Your workbench
RTK f/w 3.10
Facet
Teraterm on Windows

### Steps to reproduce
- Enter serial menu.
- choose Configure Base
- toggle base mote to get Fixed/Static position
- toggle coord system to get Geodetic
- choose Set Lat/Long/Altitude
- copy decimal degree latitude from notepad  (eg. 40.123
- paste into teraterm window
- things get very bonky.  Numbers I pasted in do not show.  weirdness ensues.
-

### Expected behavior
pasting numbers should act just like manually typing them in.

nseidle, 2023-12-12:
I *think* this is an issue with TeraTerm. You have to right click to paste (not ctrl+p). 

I am able to copy/paste from notpad using this technique. 

Let me know if this doesn't fix your problem.

closed 2023-12-12 by nseidle

tonycanike, 2023-12-12:
Oh!  Perhaps I've never pasted with TeraTerm before.  Thanks for the response, I'll give it a try.