sparkfun/SparkFun_RTK_Firmware #736: AP Config menu - add function to force update of L-Band Keys (perhaps on next reboot?)

tonycanike, 2023-12-12:
### Subject of the issue
Had to connect to PC and use the serial menu to force L-Band Key Update (Main->P->3)

Is is possible to add a function to update L-Band keys in the AP Configuration interface, or to force an update on the next reboot?
Or perhaps if the keys are expired, it should try to connect to WiFI and update the L-Band keys on every reboot?

According to the tool tip in the AP Config menu, it only tries once a day; I'm leaving on a trip and couldn't wait until tomorrow.

I presume when the Facet is in WiFi AP Config mode it can not simultaneously use WiFi to update keys, but perhaps it could do so on the next reboot. 

### Your workbench
FW 3.10

### Steps to reproduce
1. Work in multiple offices around the East Coast with different WiFi networks.
2. Turn on L-Band Facet in Office 1.  Configure with WiFi information for Office 1.
3. Keys update automagically!  Wonderful!
4. Use L-Band Facet successfully for a week or two in the field.  Lots of L-Band love!   Probably never turned it on in the office.
5. Leave it turned off for a few weeks.
6. Travel to Office 2.
7. Turn on L-Band Facet.
8. Watch WiFi connection fail (it doesn't know WiFi info for Office 2).  See L-band key update fail.  
9. Push-button to config, connect to RTK AP and rtk.local
10. Enter WiFi information for Office 2 in WiFi slot 2.
11. Save and Exit.
12. Watch Facet reboot and not update keys.  See "Keys Expired" screen.  See "LBand Config" screen.  See screen flash something.  
13. Facet goes into Rover mode.  
14. Put facet in Config mode.
15. Connect to AP.
16. Open PP config menu - see keys expire in "-2" days.
16. Go to WiFi config and put current (Office 2) WiFi info in WiFi slot 1.  (Docs say it connects to the "Home" wifi - is that slot 1 only?)
17. Save, exit, reboot, see "Keys Expired", "LBand Config", screen flash, Rover mode.
18. Put Facet into Config, connect to AP, Open PP config, click on Tool Tip - the Tool Tip says it only tries to update once a day. Sadness blankets the land.
19. Power cycle again.  See Keys Expired message again.
20. I'm leaving in the early morning to work for a week in lousy internet and lousy wifi land, so gotta fix this now.
21. USB to PC, Tera Term, Serial Config, Main->P->3.
22.  Looks like keys updated successfully.
23. Powercycle.
24. See 61 days until PP keys expire, screen flash, Rover.  Yay!
25. Powercycle again and see 61 days until expiry again.  All looks great!
26.

nseidle, 2023-12-13:
Someone else asked for nearly exactly this just this morning. Crazy.

So through the magic of television:

[image]

Yes, we *could* add a button to the WiFi config page, but I think it's worth a main display menu for ease of use.

* If no WiFi networks are entered into the settings, a "No SSIDs" message is displayed
* If networkCount > 0, device attempts to connect to local wifi. If not matching SSIDs are found, "No WiFi" is displayed.
* If WiFi is available, keys are then updated.

The RC binaries are in. Please give Dec 13 or later a try.

tonycanike, 2023-12-13:
I think that's a great decision.  Easy-peasy.  Awesome!

nseidle, 2023-12-27:
Added in [v4.0](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v4.0).

closed 2023-12-27 by nseidle