sparkfun/SparkFun_RTK_Firmware #741: GNSS RTK Facet hard locks, unable to power off using button push

bendauphinee, 2023-12-23:
### Subject of the issue
After attempting to change a config, my Facet hard locked up, and will not respond to power button press or hold (greater than 10 seconds). I think at this point I'm going to have to let its battery die.

### Your workbench
* What version of RTK firmware are you running? 3.10
* What radios are you using: WiFi. Browser access to config via iPhone / config mode.

### Steps to reproduce
I don't know that I have a reproducible one, but here's what I did leading up to this lockup.
* Added the AP name / password for my iPhone hotspot mode.
* Set the Facet to AP mode.
* Instead of connecting to my iPhone, it connected to my previously configured home AP.
* Opened a browser to Facet, changed configuration to remove home AP. Save/reset.
* Put Facet back to config mode. Fails to connect to iPhone, shows SSID/IP display, with blank SSID and 192.168.4.1 address.
* Ok, turn off iPhone hotspot, connect to Facet via it's Wifi AP.
* Removed iPhone hotspot from Wifi config, added back in home AP.
* Save settings.
* Lockup
* Try to power cycle, no response. Currently still on SSID/IP display, with blank SSID and 192.168.4.1 address.

### Expected behavior
Settings should be saved, or at least device should be able to power cycle via power button hard press.

bendauphinee, 2023-12-23:
Update: So after cracking it open, and resetting using the internal button, it looks like the issue is more awesome than that. Rebooting, and going to config mode attempts to connect to my iPhone AP again, fails, and ends up broadcasting the Rover Config Wifi. Connecting to that allows me to see settings, but modification doesn't seem to work at all.

So, reproduction steps, maybe just set it into WiFi mode for configure, give it an AP that doesn't exist, and reboot into config mode again?

bendauphinee, 2023-12-23:
Followed the [factory reset steps](https://docs.sparkfun.com/SparkFun_RTK_Firmware/configure_with_settings_file/#forcing-a-factory-reset), after powering it down using the internal button. Device is now recovered into a working state.

nseidle, 2023-12-27:
Sorry for the hassle! Your issue may be loosely related to [this forum topic](https://forum.sparkfun.com/viewtopic.php?f=116&t=60707) (this is just a breadcrumb for me).

A factory reset is a fine first step. Glad to hear you are *almost* back up.

> my Facet hard locked up, and will not respond to power button press or hold

This would indicate the external power button has failed. After opening up the unit, is your external power button working again? 

Your steps to recreate are good! Thank you. But I have a few clarifying questions.

> Added the AP name / password for my iPhone hotspot mode.
> Set the Facet to AP mode.
> Instead of connecting to my iPhone, it connected to my previously configured home AP.

How did you set the Facet to AP mode? Via serial config? Or are you saying that you pressed the setup button until the 'CONFIG' menu was selected and the device entered config mode? 

Was the device's [**Configure Mode** setting](https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_wifi/#configure-mode-ap-vs-wifi) set to AP or WiFi? I realize this is a potentially very confusing question, but it's important to be sure.

> Rebooting, and going to config mode attempts to connect to my iPhone AP again, fails, and ends up broadcasting the Rover Config Wifi. 

This is expected behavior if the device simply cannot connect to a given WiFi network. In this case, it's your iPhone. Perhaps the credentials were slightly incorrect. Can the RTK Facet connect to any other WiFi network?

It sounds like you have an SD card being used. If a unit detects an SD card, it will give precedence to the settings on the card as opposed to what is stored in the unit.

nseidle, 2023-12-27:
> Added the AP name / password for my iPhone hotspot mode.

How many WiFi networks did you have inputted into the RTK Facet?

nseidle, 2023-12-27:
I may have found the start of the problem: If I enter bad WiFi credentials, the device will fail to connect to the given SSID (to be expected). It is then designed to fall back to AP mode, but it does not:

> Connecting WiFi... No friendly WiFi networks detected.
> Connecting WiFi... No friendly WiFi networks detected.
> wifiStartAP: Error setting WiFi protocols: ESP_ERR_INVALID_ARG
> WiFi AP Started with IP: 192.168.4.1

This error is related to bad protocol switching. I'll try to pin it down more.

Potentially [related issue](https://github.com/sparkfun/SparkFun_RTK_Firmware/issues/692) (just more breadcrumbs to myself).

nseidle, 2023-12-27:
Ok, I think I found a few issues:

* The SSID 'RTK Config' would fail to print when the device fell back from WiFi config mode to AP config mode. This 'fall back' happens when either there are no friendly SSIDs detected, or a friendly SSID failed to connect (almost always because the password is bad).
* WiFi protocols were not getting set correctly because of an [improper esp_wifi_set_protocol call](https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/04eef228f49ca55ffe5d758696d8c1e0a840ec38) in AP mode.

Please give the Dec 27 release candidate a try. I believe these fixes should solve your issue.

You may still have an external button fail/problem. That's a hardware issue that we can solve with parts. Let me know if you're still experiencing an external button failure.

bendauphinee, 2023-12-27:
> How did you set the Facet to AP mode? Via serial config? Or are you saying that you pressed the setup button until the 'CONFIG' menu was selected and the device entered config mode?

Via Config mode, and then in the web interface, yes. I've yet to start tinkering under the hood with the serial mode.

> Was the device's [Configure Mode setting](https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_wifi/#configure-mode-ap-vs-wifi) set to AP or WiFi? I realize this is a potentially very confusing question, but it's important to be sure.

Shoot, I should have noted that. I was in AP mode, but I _think_ I had swapped back to WiFi mode before I removed my home WiFi settings.

---

After factory reset, the external power button worked as expected. To power cycle the device to be able to run the factory reset via SD card, I had to access the internal power button, which leads me to think that maybe future Facets need to have a design tweak to expose the internal power button as a reset button or something.

nseidle, 2023-12-27:
Ok, I think I've got this fixed in [v4.0](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v4.0). I'm going to close this issue but please feel free to open another if you see anything new/different/again.

> maybe future Facets need to have a design tweak to expose the internal power button as a reset button

Noted. Thanks!

closed 2023-12-27 by nseidle