sparkfun/SparkFun_RTK_Firmware #748: WIFI CONNECTION PROBLEM - RTK FACET GN RTK RECEIVER

Raulricardo23, 2024-01-04:
Lately I have had a problem trying to connect my Base to my cell phone.

1.- I put my Base so that everything is positioned globally well up to that point.

2.- The problem is when I put it in CONFIG mode to be able to configure it from my cell phone, it doesn't let me log in, it disconnects.
    - it connects but after a short while it takes me out of the wifi and I connect it again
    (the base equipment screen is in config mode) and I just can't
     enter via cell phone. connects and disconnects.
- It connects sometimes and everything is fine and when I want to save the changes I see that it does not save and I bother the computer and it restarts for no apparent reason.
- other times I connect it to my cell phone and it disconnects and I say why and I turn to the computer and it restarts without me doing anything.

-as the device is connected and I get on the cell phone to enter and I do not detect a signal and the device restarts itself without being able to access the Wi-Fi

3.- Lately I have installed the firmware to see if this will remove this annoying issue, but it is getting worse and worse with each update that is made.

Does anyone know the solution to this problem.

EQUIPMENT: RTK FACET GN RTK RECEIVER
                 FIRMWARE 5.0 INSTALLED

CELL PHONE: OPPO reno 6

Raulricardo23, 2024-01-05:
testing, I'll comment on the result

Raulricardo23, 2024-01-08:
Whatever you do, update the device to firmware v4.1RC

After I saw that it worked well on my equipment (Base+Radio Telemetry 915 Mhz100 mW) I could configure the fixed base or survey, it saved the changes, it turned on and off, it connected, everything was fine.

Then I installed the firmware on my other device (Rover+Radio Telemetry 915 Mhz100 mW) and I wanted to connect it and it would log in and then while on the cell phone, it would exit and the device would restart and it would not connect to the cell phone.

and I looked at what configuration was on the other computer and I saw that it had MDNS ACTIVATED, so I left it the same as the other computer that worked well, and until now everything works well with the 4.1RC update and activating MDNS, attached photo of what I have configured
[image]

nseidle, 2024-01-25:
Very good! It sounds like you're up and running! Please feel free to open a new issue if something else needs attention.

closed 2024-01-25 by nseidle

Raulricardo23, 2024-01-25:
Apparently the fix was temporary, it fails, what I was observing as a failure.....

If I have the device configured in fixed mode, if I enter from the cell phone that is when it fails and the device restarts and it does not let me change it to fixed mode, and I tried with other cell phone brands such as Samsung and nothing, then the solution it leaves me with is.
With the computer I log in via Wi-Fi and I set it to fixed mode and it lets me log in and it works wonderfully.

"then the problem I DETECT is the following:"
1. If you have the device in fixed mode, when you want to enter to configure it, it locks and does not let you enter, and the device restarts and the cell phone disconnects.
    The solution I saw is to connect from the laptop and it works wonderfully, otherwise with the cell phone

2. If it is not in fixed mode, the rest works normally, the problem is when you leave it in that fixed mode, it does not enter with the cell phone