sparkfun/SparkFun_RTK_Firmware #755: In RTK base station mode, support Trimble systems with RTCM 1008 message.

magnuan, 2024-02-01:
### Subject of the issue
In order to use the device as a base station for Trimble, it needs to transmit a RTCM 1008 message (Antenna descriptor)

Best would of-course be to be able to configure real antenna parameters, but just to make Trimble accept the data stream a 0-filled descriptor aka  ADVNULLANTENNA should be sufficient.

There is an example in this [forum](https://www.thecombineforum.com/threads/how-to-use-zed-f9p-as-a-base-station-for-trimble.331721/?sortby=newest#replies), with some python code that does this.
It just inserts a predefined 1008 message (filled with zeros)  into the stream after each 1005 message.
[0xd3,0x00,0x06,0x3f,0x00,0x00,0x00,0x00,0x00,0x99,0x25,0xca]

nseidle, 2024-02-01:
This seems very doable. I can't give you a timeline but we'll get it onto our roadmap.

ennu, 2024-06-16:
> This seems very doable. I can't give you a timeline but we'll get it onto our roadmap.

Due to local (Caribbean Island) breakdown of only exisiting Trimble base station the need for transmit a RTCM 1008 message (Antenna descriptor) on sparkfun RTK base station is high.

Any new timeline available?