sparkfun/SparkFun_RTK_Firmware #757: UBX Checksum errors are back

Stefal, 2024-02-19:
### Subject of the issue
I get some checksum errors in ubx file

### Your workbench
* What version of RTK firmware are you running? v4.0
* What radios are you using: Bluetooth

### Steps to reproduce
Mode : Rover connected to a base station via Ntrip (Bluetooth)
Frequency : 5HZ
Nmea : default settings
ubx : NAV_HPPOSLLH, NAV_PVT, NAV_SAT, NAV_TIMEGPS, NAV_TIMEUTC

Results with UBX Integrity Checker : 
```
UBX Sync Char 1 (0xB5) found at byte 42559775. Checking for Sync Char 2
NMEA $ found at byte 42560419. Attempting to process the message
Panic!! Excessive NMEA message length!
Sync lost at byte 43492973. Attemting to re-sync
Panic! Maximum rewind attempts reached! Aborting...

Processed 43492975 bytes
File size was 70826020
FILE SIZE MISMATCH!!
Longest valid UBX message was 652 bytes
Longest valid NMEA message was 85 characters
Message types and totals were:
Message type: GNTXT   Total: 6
Message type: 0x01 0x07   Total: 37389
Message type: 0x01 0x14   Total: 37396
Message type: 0x01 0x35   Total: 37386
Message type: 0x01 0x20   Total: 37430
Message type: 0x01 0x21   Total: 37429
Message type: GNRMC   Total: 7471
Message type: GNGGA   Total: 7472
Message type: GNGSA   Total: 37378
Message type: GPGSV   Total: 53888
Message type: GLGSV   Total: 39508
Message type: GAGSV   Total: 39234
Message type: GBGSV   Total: 36150
Message type: GQGSV   Total: 7495
Message type: GNGST   Total: 7495
Number of successful resyncs: 101
```

Eric-FR, 2025-05-11:
Same for me.
RTK Express.

$GNTXT,01,01,00,ESP_RST_POWERON*1C
$GNTXT,01,01,03,vv4.1*79
$GNTXT,01,01,04,HPG 1.50*30
$GNTXT,01,01,11,1A4D52539F*2F
$GNTXT,01,01,07,181E*2B
$GNTXT,01,01,09,250511,103718*7A

Panic!! UBX checksum error!
Sync lost at byte 1331000. Attemting to re-sync.

Sync has been lost. Currently processing byte 1331000. Rewinding to byte 1330958

NMEA $ found at byte 1330960

Sync successfully re-established at byte 1331033. The NMEA message started at byte 1330960

Panic!! UBX checksum error!
Sync lost at byte 2297093. Attemting to re-sync.

Sync has been lost. Currently processing byte 2297093. Rewinding to byte 2295475

UBX Sync Char 1 (0xB5) found at byte 2295533
[...]
NMEA $ found at byte 24265184

Sync successfully re-established at byte 24265239. The NMEA message started at byte 24265184

Processed 32596657 bytes
File size was 32596657
Longest valid UBX message was 1912 bytes
Longest valid NMEA message was 85 characters
Message types and totals were:
Message type: GNTXT      Total: 6
Message type: GNGST      Total: 11901
Message type: 0x02 0x15  Total: 11875
Message type: GNRMC      Total: 11915
Message type: GNGGA      Total: 11915
Message type: GNGSA      Total: 59557
Message type: GPGSV      Total: 17906
Message type: GLGSV      Total: 15614
Message type: GAGSV      Total: 15161
Message type: GBGSV      Total: 16504
Message type: GQGSV      Total: 2979
Message type: 0x02 0x13  Total: 67098
Number of successful resyncs: 13

Bye!

Eric-FR, 2025-10-07:
Still waiting for firmware 4.3. Lost in translation???

PaulZC, 2025-10-08:
PS. @Eric-FR : there is a Release Candidate available which includes these improvements: 

https://github.com/sparkfun/SparkFun_RTK_Firmware_Binaries/blob/main/RTK_Surveyor_Firmware_RC-Sep_23_2025.bin

https://github.com/sparkfun/SparkFun_RTK_Firmware_Binaries