sparkfun/SparkFun_RTK_Firmware #763: BLUETOOTH_NO-NMEA-MESSAGES

kmnkpy, 2024-04-23:
### Subject of the issue
The device RTK_FACET stop sending out NMEA-MESSAGES through Bluetooth after a minute when connected to phone. The serial port through Bluetooth is working. I have to switch off the device and wait for long time to get NMEA through Bluetooth. I am getting. 
NMEA via usb comport without any interrupt.

### Your workbench
RTK-firmware ver-4.0
Using SW maps and serial Bluetooth terminal for android 
After powering the device, it shows serial config in the screen, then it goes when I connect it the serial Bluetooth terminal app and enter the configuration menu and press x. if I disconnect the serial terminal app the serial confiig screen displays again.
### Steps to reproduce
Power on device and connect to serial terminal or SW map and wait minute.
### Expected behaviour
it should give the location data without any interrupt and no serial config screen.

### Actual behaviour
The device RTK_FACET stop sending out NMEA-MESSAGES through Bluetooth after a minute when connected to phone.

nseidle, 2024-04-23:
> when I connect it the serial Bluetooth terminal app and enter the configuration menu and press x. if I disconnect the serial terminal app the serial confiig screen displays again.

This is very odd. The Serial Config screen should only appear if an external character is received. When you power on the device,  does the 'Serial Config' screen appear even if a USB cable is NOT connected? If yes, then we may need you to [return the unit](https://www.sparkfun.com/returns) for inspection.

> Using SW maps and serial Bluetooth terminal for android

I can confirm that the Serial Bluetooth Terminal will freeze after 16 to 17 seconds. I suspect this has something to do with that app and the shear amount of data it needs to buffer. 

However, using SW Maps, I can stay connected for over 30 minutes (and I assume for hours) without an issue.

Do you experience this issue using SW Maps?

Have you increased the [measurement frequency](https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_gnss/#measurement-frequency) beyond 4Hz?

kmnkpy, 2024-04-24:
> This is very odd. The Serial Config screen should only appear if an external character is received. When you power on the device, does the 'Serial Config' screen appear even if a USB cable is NOT [...]

yes, the serial config menu appears when I power on the device. The device was returned to sparkfun for inspection in January last (RMA 150239) and I received it repaired on this month and when I received the unit it was ok there. after 3 days it started to show this problem.

closed 2024-04-24 by nseidle

kmnkpy, 2024-04-24:
The gnss is working and I am getting rtk fix using SW maps and usb c port connectivity. But the life isn't nice without Bluetooth tech. The part was purchased through Mouser electronics and the rma number given by them. The concerned persons are Venu from Mouser and john Trepke from sparkfun. For me the return is time consuming and costly since I am outside US (India).

nseidle, 2024-04-24:
If all you need is Bluetooth and GNSS, you are welcome to [downgrade the firmware](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#updating-using-windows-gui) (you cannot do harm to the unit) and see if the behavior changes. I cannot explain why Bluetooth is stopping. If you'd like to troubleshoot it, please capture the serial output from Power On and post the contents here. If you change the firmware, please let me know.

Also, consider starting from a [Factory Reset](https://docs.sparkfun.com/SparkFun_RTK_Firmware/menu_system/#factory-reset) after each firmware upgrade/downgrade.

kmnkpy, 2024-04-24:
it is not the Bluetooth which stopping instead the NMEA messages are stopped over Bluetooth. I can still connect the instrument via Bluetooth serial port and issue serial command I even issued factory reset from the phone using Bluetooth serial comport app and it worked.

kmnkpy, 2024-04-24:
How to capture serial output from Power On.

nseidle, 2024-04-24:
Ah, that's interesting. When you connect to the device over Bluetooth, you should see NMEA streaming. Pressing `+++` should enter [Bluetooth Configure](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/configure_with_bluetooth/) mode. Is that what you are experiencing? Or is the unit already in configure mode when you open the Bluetooth Serial Terminal app?

> How to capture serial output from Power On.

We use [TeraTerm for Windows](https://learn.sparkfun.com/tutorials/terminal-basics/tera-term-windows). From here we can save the output to a log file, or simply copy/paste the output:
[image]

kmnkpy, 2024-04-25:
I am attaching log file. The port used is CH34
[teraterm.log](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/15108568/teraterm.log)

kmnkpy, 2024-04-25:
> Ah, that's interesting. When you connect to the device over Bluetooth, you should see NMEA streaming. Pressing `+++` should enter [Bluetooth [...]

yes, when I connect, I see NMEA after some time it stops. After that pressing +++ give me configure menu.

nseidle, 2024-04-25:
Thank you but the log doesn't show a device startup. Please start logging data from TeraTerm, then issue the letter **s** (system menu), **d** (debug), **r** (restart). Once it's fully started, please post the log.

kmnkpy, 2024-04-25:
> Thank you but the log doesn't show a device startup. Please start logging data from TeraTerm, then issue the letter **s** (system menu), **d** (debug), **r** (restart). Once it's fully started, please post the log.

please see this log 
[log.txt](https://github.com/sparkfun/SparkFun_RTK_Firmware/files/15113355/log.txt)

nseidle, 2024-04-25:
entry 0x400805f8
    0x36 - MAX17048 Fuel Gauge
    0x38
    0x39
    0x3a
    0x3d - SSD1306 (64x48) OLED Driver
    0x42 - u-blox ZED-F9P GNSS Receiver
    0x44
    Display started

Something is very odd with your unit. Those are I2C devices appearing on the I2C bus and they do not exist (0x38, 0x39, etc are not known or valid I2C addresses). 

Please open an RMA and reference this issue. I'll work with customer service to get your unit replaced.

kmnkpy, 2024-04-26:
Oh, that is disappointing. I have purchased sprkfun reference station and facet. The reference station did not show any issue until now. But facet I purchased shown issue and returned for RMA and came back repaired, again I have to return, that is the disappointing part. In both of them facet I liked.

nseidle, 2024-04-26:
I am sorry for the hassle. Please contact [tech support](https://www.sparkfun.com/returns) and we'll get it fixed.

kmnkpy, 2024-05-31:
Hello Nathan.

I received a new unit from sparkfun as replacement for the faulty unit.
Thanks for your attention to this matter.

Thankfully
Naseer km.