sparkfun/SparkFun_RTK_Firmware #770: No NMEA over Bluetooth

TheCre8r, 2024-05-10:
### No NMEA over Bluetooth
Same issue as #665 Works with version 1.6 no later versions

### Your workbench
* Version 1.6 is the latest working version to receive NMEA stream over Bluetooth.

### Steps to reproduce
Testing using COM over Bluetooth with Putty.

Tersted by clearing the flash and going straight to newer versions, also tested incremental updates.

### Expected behavior
Receive NMEA stream over Bluetooth COM.

### Actual behavior
Did not receive recieve NMEA stream over Bluetooth COM.

nseidle, 2024-05-20:
Thanks for reporting but I'm still not able to replicate. I'm using v4.0 below:

[image]

[image]

[image]

Above, the device is seen and connected to over Bluetooth. 

Bluetooth is the primary way users connect to our devices to get NMEA and we haven't heard any other issues so I don't suspect it's a firmware issue. But then you say v1.6 is working for you. If v1.6 is working for you, please continue to use it.

closed 2024-05-20 by nseidle

tonycanike, 2024-05-20:
I'm finally getting around to upgrading from f/w 3.10 to 4.0.   I'm running SurvPC 7.0 on a Panasonic Toughpad running W10.

I connected SurvPC to my Facet f/w 3.10 via Bluetooth.  SurvPC successfully received and parsed the NMEA data.

I upgraded my Facet to f/w 4.0 via the WiFi config - the upgrade worked flawlessly, very nice.

I connected SurvPC to my Facet f/w 4.0 via Bluetooth.  Again, SurvPC successfully received and parsed the NMEA data.