sparkfun/SparkFun_RTK_Firmware #771: Failure in Wifi Config when using base coordinates saved in SD in GGG MM SS.ssss format

amlago, 2024-05-23:
Failure in Wifi Config when using base coordinates saved in SD in GGG MM SS.ssss format

When trying to add coordinates previously saved in the base coordinates file in GGG MM SS.sssss format, they are copied to the boxes incorrectly.
The Facet detects the spaces between GGG and MM and SS.ssss and copies the degrees in latitude, the minutes in longitude and thus fills in boxes.
I am on version 3.8, which is the one that works best for me in my activity, where I almost always use the base Facet.

amlago, 2024-05-23:
This started a long time ago with entry #428.

PaulZC, 2024-06-05:
Hi Angel (@amlago ),

Thank you for reporting this issue.

The root cause of this is ```loadGeodetic``` in ```main.js``` ( [link](https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/ad3899b609e2d2b3f88d74b7b3c3411c0b72834f/Firmware/RTK_Surveyor/AP-Config/src/main.js#L1370-L1375) ). It uses ```split``` to split the station coordinates into parts. The problem is that this was written for the ```DD.ddddd``` format, and fails badly with the ```DD MM SS.sss``` format. The extra spaces cause the latitude MM to be treated as longitude.

I will work on a fix for this.

Best wishes,
Paul

PaulZC, 2024-06-06:
This has been corrected - in commit https://github.com/sparkfun/SparkFun_RTK_Firmware/commit/1675bb4329cf5ba4821c659525a56d44887e9c1d

There will be a RC binary available for test in 1-2 days

PaulZC, 2024-06-06:
Also related:

```DD-MM-SS.ssss``` was being mistaken as ```DD-MM-SS``` by ```coordinateIdentifyInputType```. This led to rounding errors after the base coordinates had been saved / entered.

Resolved in commit 3cf5141e806addf9810e8da85be820f0bc4f8aeb

amlago, 2024-06-06:
excellent. I'm waiting for the RC and I'll try it.
Greetings

amlago, 2024-06-13:
I will try it and make the corresponding comments.
My greetings
Angel

amlago, 2024-06-13:
Entering coordinates in other formats seems to work fine. I will inform you of any details if necessary.
Thank you

nseidle, 2024-07-22:
Fixed in [v4.1](https://github.com/sparkfun/SparkFun_RTK_Firmware/releases/tag/v4.1).

closed 2024-07-22 by nseidle