sparkfun/SparkFun_RTK_Firmware #772: Support for ESRI Field Maps on iOS

DShaeffer, 2024-05-25:
I work for the U.S. Army, and we've purchased both versions of the Facet for testing. Unfortunately, ESRI Field Maps, Survey123, and Quick Capture do not work with the RTK Facet and RTK Facet L-band via BLE on iOS. These are industry-standard mobile apps for collecting GPS data in the environmental field within the United States. 

The units work great with Android, and we are otherwise very satisfied with them. However, many of our users prefer iPhones. We would likely purchase many more units if they could be made to work with the three aforementioned mobile apps. 

Is it possible to adjust the firmware to support these apps or is this a hardware issue?

gdt, 2024-05-25:
You should be aware (and probably are, but better to be sure) that there is Bluetooth Serial Port Profile (often called classic bluetooth), and BLE.   As I understand it somewhat fuzzily, Apple has some sort of licensing scheme for regular bluetooth, where device manufacturers are expected to pay in order that iUsers can use them, despite bluetooth being a standard.   IMHO this stance by apple is the root of much trouble.

I have only tried to use Android (which does support Bluetooth SPP), but the docs say the firmware supports BLE.  Surely you aren't the first to want this.    Have you configured it to be BLE instead of SPP?  Have you been able to make BLE work with Android   Are you running the latest firmware?

Thank you for being clear about your affiiliation.   I am in the US, and thus have no ITAR concerns about helping you, but people in other countries may well run afoul of doing so without a license, assuming harmonized regulations.

tonycanike, 2024-05-25:
My understanding has been that this is an issue with the applications not supporting the bluetooth connection. 

Here's a screen shot from the QField help pages - it says QField does not support Bluetooth on iOS. 

[image]

So I would say this is a question for the app developers, not for SparkFun.

DShaeffer, 2024-05-25:
I am using the most recent software release (v4.0). I understand that Apple has a unique licensing scheme for Bluetooth and only supports BLE. This is one of the reasons I avoid iOS, as such decisions limit its usefulness.

I have successfully connected to the Facet in Field Maps via SPP on Android. I can connect to the Facet using BLE on Android and the Facet as a location provider in Field Maps, but it doesn't provide location any data (says cannot connect to GPS). On iOS, I cannot even see the BLE connection. I'm not familiar with QField, but Tony's comment might have some validity. There isn't a way to initiate a BLE connection via Field Maps. However, ESRI Quick Capture can initiate a BLE connection in the app, but the Facet still doesn't appear in the device list. I can certainly reach out to the software vendor, but it would be helpful if others could download these free ESRI apps and see if they have any success connecting.

gdt, 2024-05-25:
As an aside, QField is the most popular mobile app companion for QGIS.  It works pretty well, but if you aren't using QGIS it's a lot of effort to set up for it.  And, for now it doesn't support BLE on iOS, although I'd hope it's a small matter of programming away.

Are you able to connect to any other GNSS receiver (which supports BLE) from these apps in iOS?   There are two thing to sort out here: 1) whether the apps can do BLE correctly and 2) whether the SF implementation of BLE is correct and works with a correct app.  Right now neither of those is clear.

It sounds like you are saying that even after reconfiguring the SF to BLE, and using an Android app, you can't get working location.  Does that app and device work with any other GNSS receiver over BLE?

All of this has me wondering if "serial port over BLE" is basically somewhat troubled.

There is also a TCP server, where the SF device can listen, I think on 2947.  I have used this scheme with an Ardusimpler receiver, to use it from Vespucci, which didn't have any bluetooth support.    But perhaps apple doesn't allow that either, and to use it generally requires putting the SF on a hotspot created by the phone, which might raise other issues.  It seems to be a common workaround for bluetooth issues.

Probably you could bring an actual serial port, or serial over USB out,  and connect that way.   But perhaps apple doesn't allow that either.

nseidle, 2024-05-29:
@DShaeffer - I hear what you're saying; you just want it to work. We are likely not currently compatible with Esri apps. It is *not* a hardware issue, it's likely a limitation of the firmware. We'll add it to the top of our work lists.

DShaeffer, 2024-05-29:
@needle. I haven't had much success with using BLE on Android. I was able to connect to the receiver and view the NMEA sentences using the Android BLE Scanner app, but I couldn't get Android SW Maps to work in BLE mode. I could connect to the receiver on my phone and see it in SW Maps, but when I tried to connect, it would say the receiver is disconnected. Bluetooth Classic works great on Android, though. I should probably stop trying to cater to users who insist on using iPhones. As an electronics hobbyist, I've learned a lot from Sparkfun and Adafruit hardware. I'm trying to encourage our agency to adopt more open-source hardware instead of Trimble receivers.

I will post on the ESRI Early Adopter forums to see if I can get a response regarding BLE.

nseidle, 2024-06-10:
@DShaeffer - We are able to get Survey123 and QField working great on iOS. We first made it work on the newer RTK devices so the docs are [here](https://docs.sparkfun.com/SparkFun_RTK_Everywhere_Firmware/gis_software_ios/) until I can get them moved over to this repo. QuickCapture looks like it will work just like Survey123.

ESRI Field Maps is oddly different and doesn't show a network option (TCP).  Even more so, the Windows version seems to have a TCP option (but is being [deprecated in 2025](https://doc.arcgis.com/en/field-maps/get-started/pdf/ConnectFMWindowsToReceiver.pdf)). Also, there are other folks having similar questions ([Emlid RS2](https://community.esri.com/t5/arcgis-field-maps-questions/tcp-connection/td-p/1317657)). Very odd.

DShaefferUSACE, 2024-06-10:
@nseidle  That's awesome! I appreciate you looking into that and creating documentation. I've posted about the lack of support for BLE/TCP in the Field Maps app on ESRI's "Early Adopters" forum, as the development team tends to monitor it more closely and respond more quickly than the public forum you linked. If I don't get an answer there, I'll reach out to our dedicated rep. Unfortunately, ESRI is a large company and can be slow to implement new functionality, even if they like the idea.

nseidle, 2024-06-11:
The docs have been [updated](https://docs.sparkfun.com/SparkFun_RTK_Firmware/gis_software_ios/#arcgis-survey123). Please give them a go and let us know if anything doesn't make sense.

closed 2024-06-11 by nseidle