sparkfun/SparkFun_RTK_Firmware #776: Facet hung issuing corrections to Rtk2go and Emlid simultaneously

amlago, 2024-06-15:
Facet hung issuing corrections to Rtk2go and Emlid simultaneously

facet v4.1 06/14/24
I was working today with Facet in base mode for a little over 5 hours, issuing corrections to Rtk2go and Emlid at the same time. I also saved the ubx file on the SD card and when that period of 5 hours of work more or less arrived, it crashed. It stopped issuing corrections, the screen remained frozen with a fixed rtcm number without advancing.
It didn't turn off normally and I had to hold the power button for 20 seconds and it turned off.
Could it be a ram problem due to the double emission and saving of the ubx file on the SD card?
The Wi-Fi in the work area is excellent and the battery was over 60 percent.
I leave the problem to your analysis.
greetings

PaulZC, 2024-06-19:
Hi Angel (@amlago ),

Many apologies for the slow reply.

It might be a RAM problem, the double NTRIP connection does use more RAM. It will try and replicate this here, and capture the crash.

Greetings from the UK,
Paul

skyviewaerials, 2024-06-29:
I ran into the same issue on a very hot day here in Los Angeles, temps passing 100*F I removed the rubber sleeve to reduce the heat being absorbed by the Facet. I finally decided to cover the unit with a white plastic shopping bag, just covering it and poking a hole in the top to vent the heat.

amlago, 2025-04-02:
Hi, I've had the Facet freeze issue again when streaming via ntrip to two casters and saving the ubx to the SD card.
I haven't had much explanation on the issue since last year.
Has anyone had the same problem and solved it?
Regards.

PaulZC, 2025-09-10:
Hi Angel,

Apologies for the slow reply!

I am testing new Release Candidate firmware, sending corrections to both RTK2go and emlid. This is on the Reference Station, connected via Ethernet. All is working as expected so far. I will leave it running for a few hours to see if I can replicate your issue.

Best wishes,
Paul