sparkfun/SparkFun_RTK_Firmware #778: LIS2DH12 still not used?

sixtemesseven, 2024-07-27:
Hi

Even though the Facet does have an Accelerometer, tilt compensation is not implemented.
I read an earlier disscusion and you wrote you were working on the math.

The torch has it, but as I understand the compensation is done directly in the GPS module.

Really happy with the Facett, but tilt would be great!

Any updates on this?
Thank you.

amlago, 2024-08-06:
I'm very interested if this function could be implemented.
Does the hardware allow for data to be obtained to solve the math associated with this?

nseidle, 2024-08-06:
> The torch has it, but as I understand the compensation is done directly in the GPS module.

The Torch uses the [IM19](https://cdn.sparkfun.com/assets/2/2/d/f/d/IM19_IMU_Tilt_Sensor_v1_1_10.pdf), a sophisticated module specifically built for GNSS tilt compensation to augment the UM980 GNSS receiver. 

I'm sorry but tilt compensation is very complex and requires additional hardware (namely, gyros and temperature compensation) that we will not be implementing on this line of products. If u-blox or other GNSS model manufacturers add tilt compensation as part of their receiver (ie the ZED-F9R with a surveyor model as opposed to an automotive model they use today), we can look to add tilt compensation to future products.

closed 2024-08-06 by nseidle