sparkfun/SparkFun_RTK_Firmware #779: Error entering base coordinates in serial terminal

amlago, 2024-08-29:
Error entering base coordinates in serial terminal.

Facet v4.1
Using the serial terminal program, the base coordinates cannot be entered.
The user enters the coordinates but they do not change on the screen or are recorded in memory. I tried various formats but they all fail.
From the wifi configuration, the base coordinates can be changed and it works fine.
Regards

PaulZC, 2024-08-30:
Hi Angel (@amlago ),

I have tested this using different formats and it is working OK for me:

[image]

[image]

Can you please share some screenshots?

Thank you,
Paul

amlago, 2024-08-30:
Hi Paul.
What a strange thing that's happening to my Facet.
I'll try entering coordinates again later and send you the screenshots.
As I told you, everything works fine for me in the wifi configuration.
See you soon.
Regards

amlago, 2024-08-30:
Hi Paul. Everything works fine.
I was making a mistake when entering the data.
I was entering the lat, long and height coordinates all together separated by commas.
They are entered one by one. Something I already knew but I had a slow thinking day.
Thank you very much for answering and apologies.
greetings
Angel

PaulZC, 2024-08-30:
No problem Angel,
Have a nice afternoon,
Paul

closed 2024-08-30 by PaulZC