sparkfun/SparkFun_RTK_Firmware #781: Accessing base coordinates saved in SD by serial bluetooth configuration

amlago, 2024-10-16:
Accessing base coordinates saved in SD by serial bluetooth configuration

I am interested in configuring the Facet in base mode by serial bluetooth configuration using coordinates saved in the SD.
I already do that from wifi config..
Is it possible to access the SD and the saved coordinates by serial bluetooth config?
I am interested because I have found that the serial bluetooth configuration is much more stable and secure than the wifi configuration.
Sometimes the wifi config does not save the changes or I have to refresh the web page for it to do so.
The changes by serial bluetooth config are instantaneous and secure.
I appreciate the comments.
greetings

Angel

nseidle, 2024-10-25:
Hello @amlago!

What setting are you trying to change over Bluetooth serial config? 

If I understand you correctly, do you want the ability to select between different base coordinates from a list? Similar to what is presented over Web Config (shown below)?

[image]

Do you want to be able to select between stored base coordinates from the serial config menu?

tonycanike, 2024-10-25:
Angel, that's an interesting idea.  I very much like the idea of being able to add and load saved base coordinates via the serial configuration menu.  

I also sometimes find the Wifi interface unstable, or at least appear to be unstable to me. I usually load saved base coordinates twice to be sure it actually happened, and I'm never 100% confident it happened.  

The serial menu has always seemed rock stable to me.

And shouldn't the serial menu be able to do everything the Wifi configuration interface can?

So yes, being able to add (save to SD) and load (recall from SD and load into the ZED) base coordinates from the serial configuration menu would be useful.  I have never thought about doing this from the serial menu and I've never tried.  I would have guessed that functionality was already in the serial configuration menu.

amlago, 2024-10-25:
Hi Tony and Nathan.
It's a pleasure to be back to the exchange of ideas.
Regarding the wifi configuration, the same thing happens to me as Tony. I save twice and sometimes I'm not sure of the results. Sometimes I have to reload the web page so I can save new data.
The Bluetooth serial interface is not user-friendly but I'm sure of the results. It works almost 100% of the time.
It would be another interesting option for real use of the Facet in the field, being able to access and save base station coordinates via serial config.
I know they're very busy at Sparkfun but I would ask them to take some time to perfect the Facet.
It's a very versatile and practical device.
Regards
Angel