sparkfun/SparkFun_RTK_Firmware #782: Source compilation issue - missing SSLClientESP32 and mbedtls/net.h

semuadmin, 2024-11-16:
### Subject of the issue

Attempting to compile the RTK Surveyor source using the Arduino IDE in accordance with the instructions [here](https://docs.sparkfun.com/SparkFun_RTK_Firmware/firmware_update/#compiling-source). I get a series of missing library errors.

### Your workbench

Using Arduino IDE 2.3.3 on MacOS Sequoia 15.1 (ARM64).

### Steps to reproduce

1. Open the RTK Surveyor firmware source folder in Arduino IDE.
2. Set the ESP32 hardware configuration as specified in the 'compiling source' document.
3. Add all libraries listed in the 'Required Libraries' section of the 'compiling source' document via the Arduino LIbrary Manager (including those only available on GitHub).

### Expected behavior
Successful compilation and linking.

### Actual behavior

I get the following missing library error:

```
/Users/semuadmin/Library/CloudStorage/Dropbox/Development/workspace_vscode/SparkFun_RTK_Firmware/Firmware/RTK_Surveyor/RTK_Surveyor.ino:83:10: fatal error: SSLClientESP32.h: No such file or directory
   83 | #include <SSLClientESP32.h> // http://librarymanager/All#SSLClientESP32
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: SSLClientESP32.h: No such file or directory
```

On installing what I presume to be the missing library (https://github.com/alkonosst/SSLClientESP32), I get a further error:

```
In file included from /Users/semuadmin/Documents/Arduino/libraries/SSLClientESP32/src/SSLClientESP32.h:25,
                 from /Users/semuadmin/Library/CloudStorage/Dropbox/Development/workspace_vscode/SparkFun_RTK_Firmware/Firmware/RTK_Surveyor/RTK_Surveyor.ino:83:
/Users/semuadmin/Documents/Arduino/libraries/SSLClientESP32/src/ssl_lib_client.h:11:10: fatal error: mbedtls/net.h: No such file or directory
   11 | #include "mbedtls/net.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: exit status 1
```

I did a quick Google but couldn't see any obvious remedy. 

Are you able to advise on which specific SSL/TLS and/or related libraries I need to install to successfully compile the firmware?

As an aside, not an issue at all but has any consideration been given to porting this library to VSCode PlatformIO? Appreciate the Arduino IDE is ubiquitous, but the PlatformIO extension to VSCode has (imho) far superior editing and debugging facilities, and the dependent configuration and libraries can all be specified in a single `platformio.ini` file for ease of management.

gdt, 2024-11-16:
It would be nice if the README were updated so that following it was all you had to know.  Even if it says to read the yaml and get that!

I'm -1 on supporting any proprietary tooling, and more than happy to have a CLI-only build process where people are expected to use whatever editor they want to edit the sources.

semuadmin, 2024-11-16:
> It would be nice if the README were updated so that following it was all you had to know. Even if it says to read the yaml and get that!
>
> I'm -1 on supporting any proprietary tooling, and more than [...]

I have no skin in the game at this point and am probably grossly ill-informed, but that won't stop me venturing an opinion 😏 ...

Appreciate anything that impacts the current workflow is likely to be an annoyance, but structuring the repo so that's *at least* compatible with VSCode/PlatformIO (i.e. *.h and *.cpp in a /src folder, rather than using the legacy Arduino IDE's *.ino structure) - does not *compel* anyone to use VSCode (*or any other IDE*) and is, I would argue, the way C++ libraries were intended to be structured. A big plus for PlatformIO is that, as I say, it holds all the platform configuration and build deps in a single easily-digested file, which obviates (*to a large extent*) the need to maintain lengthy build instructions.

I know "real men don't use GUIs" but I'm all for making life easier for us lightweights.

I'll get my coat...

gdt, 2024-11-17:
I have no objection to making things easier for multiple toolchains and editors.   I should have said something a bit crisper, which was that I don't think it's good to favor or begin to go down the path of recommending it as a primary path.  My wording was overly enthusiastic....

nseidle, 2024-11-18:
@semuadmin - I have no problem with a restructure to support other environments. We are very much a do-ocracy. If you'd like the door a different color, please pick the color and paint it. As long as it doesn't break our current workflow (too much :)), then I'll welcome the PR.

closed 2024-11-18 by nseidle

semuadmin, 2024-11-18:
> @semuadmin - I have no problem with a restructure to support other environments. We are very much a do-ocracy. If you'd like the door a different color, please pick the color and paint it. As long as it doesn't break our current workflow (too much :)), then I'll welcome the PR.

Thanks @nseidle

As per your DM, I may focus on the newer [RTK Everywhere](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware) repo, though it'll probably be a while before I get to it :-)

Where I'm coming from is - for prospective contributors who *would* like to use an IDE for editing and *potentially* as part of the build and deploy chain, AFAIK using the Arduino *.ino structures pretty much obligates them to use the Arduino IDE (*or a handful of geriatric Eclipse plugins*), which is fine for hobbyists but not - I would argue - fit for purpose for more advanced work.

By migrating (*in due course*) to the de facto standard C++ repo structure (*.h, *.cpp in /src), the contributor is then at liberty to use any of a wide variety of IDEs (including, if they still want, the Arduino IDE), instead of or in addition to any CLI toolchains.

I personally like VSCode/PlatformIO for C++ MCU coding and deployment - *indeed, it's about the only Microsoft application I still use regularly* -  but I'm 100% behind the principle of avoiding proprietary lock-ins and unnecessary regression.