sparkfun/SparkFun_RTK_Firmware #790: Facet. Elevation mask in base mode

amlago, 2025-04-02:
Facet. Elevation mask in base mode

Hi, yesterday it seemed to me that the elevation mask defined in Facet's settings only works in rover mode.
It had no effect in base mode.
Could you please check this? I'm very interested in removing satellites with low altitude above the horizon in base mode.
Thank you.

Angel

amlago, 2025-04-03:
For some unknown reason, version 3.8 has worked better for me than the latest versions.
Of course, I can only send one caster.

PaulZC, 2025-09-09:
Hi Angel,

Apologies for the delay.

As far as I can tell, the `minElev` configuration is the same for both Rover and Base:

Rover:

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/ec065cb65d8da0d567251fb31a0e363ff0e89140/Firmware/RTK_Surveyor/Rover.ino#L94

Base:

https://github.com/sparkfun/SparkFun_RTK_Firmware/blob/ec065cb65d8da0d567251fb31a0e363ff0e89140/Firmware/RTK_Surveyor/Base.ino#L128

I did find a [small error in the GNSS library](https://github.com/sparkfun/SparkFun_u-blox_GNSS_v3/issues/86), but it should have no effect here. RTK Everywhere doesn't allow negative minimum elevations.

Best wishes,
Paul

amlago, 2025-09-10:
Hi Paul.
Thanks.
I'll check again.
Best regards
Angel