sparkfun/SparkFun_RTK_Firmware #791: Flag for clean cyclic logging files

jiargei, 2025-04-24:
### Subject of the issue

You already implemented a cyclic logging, also for long time observations (see #86 , and #43 ) which works actually fine. It would be nice to have a flag (logging to set the observation periods to propper time periods. Lets name the variable "**useCleanLogTimePeriod** (0 or 1)"

### Example Szenario

I want to use my RTK Explorer to log several days and generate hourly files (3600 s or 60 m). I start measuring at 07:49:41.. so the first file will be named 

By setting "**useCleanLogTimePeriod=0**", the already implemented behaviour will generate files:

- "SFE_Facet_yymmdd_074941.ubx"
- "SFE_Facet_yymmdd_hh4941.ubx"

By setting "**useCleanLogTimePeriod=1**", the next files would be:

- "SFE_Facet_yymmdd_074941.ubx"
- "SFE_Facet_yymmdd_080000.ubx"
- "SFE_Facet_yymmdd_hh0000.ubx"

So to check if a new file should be created, add something like "current gps time modulo maxLogLength_minutes == 0". I could'nt find the implementation in the code. 

Especially for long time observations it would be nice to have "clear" filenames and see a feature like this in some of the next releases!

nseidle, 2025-04-24:
This is a reasonable request. We're unlikely to be able to add this feature to this firmware base but I have copied your issue over to our [new code base](https://github.com/sparkfun/SparkFun_RTK_Everywhere_Firmware/issues/630).

PaulZC, 2025-09-09:
Closing... This feature is available in RTK Everywhere. I've had a good look at implementing it here, but it conflicts with logTest (`settings.runLogTest` etc.). I don't think it's possible to add it here cleanly, unless we remove logTest first.

closed 2025-09-09 by PaulZC